ahundt/grl #86: Visual Studio 2013 support

felixvd, 2015-09-08:
I am getting this error, evidently because I have VS 2013 installed. Is C++11 needed over C++12? Is there a way to try it out with 12? A Google result proposed just abandoning the version check.

     CMake Error at config/Settings.cmake:38 (message):
       The compiler C:/Program Files (x86)/Microsoft Visual Studio
       12.0/VC/bin/x86_amd64/cl.exe has no C++11 support, or our tests failed to
       detect it correctly.  Please use a different C++ compiler or report this
       problem to the developers.
     Call Stack (most recent call first):
       C:/Users/robotics/Documents/GitHub/cmake-basis/src/cmake/ProjectTools.cmake:1590 (include)
       C:/Users/robotics/Documents/GitHub/cmake-basis/src/cmake/ProjectTools.cmake:1948 (basis_initialize_settings)
       C:/Users/robotics/Documents/GitHub/cmake-basis/src/cmake/ProjectTools.cmake:2156 (basis_project_begin)
      CMakeLists.txt:84 (basis_project_impl)

ahundt, 2015-09-08:
The code requires C++11, and will probably work with later versions such as C++14. I assume that's what you meant by C++12, since there there is no C++12 standard. 

The line for the check is here:
https://github.com/ahundt/grl/blob/master/config/Settings.cmake#L30

What google result made that suggestion?

felixvd, 2015-09-08:
[This](https://github.com/RainerKuemmerle/g2o/issues/54) issue made it sound like the check might disallow later versions, so I wanted to see what compiling without the check gives.

I assume that is what I meant, if Visual Studio 2013 uses it.

ahundt, 2015-09-08:
Ok try changing the relevant C++11 lines in Settings.cmake to:

```cmake
if(NOT MSVC)

    # Enable C++11
    include(CheckCXXCompilerFlag)
    CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
    CHECK_CXX_COMPILER_FLAG("-std=c++0x" COMPILER_SUPPORTS_CXX0X)
    if(COMPILER_SUPPORTS_CXX11)
        set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
    elseif(COMPILER_SUPPORTS_CXX0X)
        set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
    else()
            message(FATAL_ERROR "The compiler ${CMAKE_CXX_COMPILER} has no C++11 support, or our tests failed to detect it correctly. Please use a different C++ compiler or report this problem to the developers.")
    endif()
endif(NOT MSVC)
```

felixvd, 2015-09-09:
CMake generated and built it fine now, and so did Visual Studio. Didn't test the resulting files, but I suppose that works.

closed 2015-09-09 by felixvd

ahundt, 2015-09-09:
cool I'll make a commit incorporating this change when I have a chance.