ahundt/grl #87: How to point to Boost directory?

felixvd, 2015-09-08:
I built all dependencies, but on Windows 8.1, CMake does not seem to find Boost out of the box:

    Looking for Boost [program_options;filesystem;unit_test_framework;system;regex;coroutine;log]...
    Could NOT find Boost
    Looking for Boost [program_options;filesystem;unit_test_framework;system;regex;coroutine;log]... - not found

No cache entry is demanded like for the other dependencies, and setting a BOOST_ROOT environment variable doesn't help. What's the right way to tell the configuration where the Boost directory is?

ahundt, 2015-09-08:
I believe it should use the default FindBoost.cmake script provided by CMake, none are provided in the source tree. The cmake scripts are in the [config folder](https://github.com/ahundt/grl/tree/master/config), plus the cmake files in the top level directory.

I've only used the library on OS X, and a little bit on Linux. I'll be happy to help you get up and running with it and apply any patches you need to add to get things working! I've been careful to use libraries that are cross platform so that shouldn't be a problem.

ahundt, 2015-09-08:
perhaps the following helps?

https://stackoverflow.com/questions/8077652/cmake-doesnt-find-boost-libraries

felixvd, 2015-09-09:
Turns out I had to set Boost_USE_STATIC_LIBS to true, otherwise FindBoost.cmake wouldn't identify the right libraries although it found the files. BOOST_ROOT or BOOSTROOT probably had to be set as well.

I note that turning "Boost_DEBUG" on as mentioned [here](http://stackoverflow.com/questions/3808775/cmake-doesnt-find-boost) was helpful as well.

closed 2015-09-09 by felixvd

ahundt, 2015-09-09:
interesting, I'm using the dynamic libraries myself