ahundt/grl #96: AzmqFlatbuffer - why define "receive_buffer_type" for received buffers? how to exploit it?

Ashkan372, 2016-06-07:
Why do we need this? 
`typedef std::shared_ptr<boost::container::static_vector<uint8_t,256>> receive_buffer_type;`        taken from here: [https://github.com/ahundt/grl/blob/master/include/grl/AzmqFlatbuffer.hpp#L41](url)

How is the buffer stored in a vector of uint8_t? 
And how should I extract the flatbufferBuilder that resides in the received buffer? (which is of the type, "receive_buffer_type")

ahundt, 2016-06-07:
Am I correct that the question behind your question is how do I receive a flatbuffer?

If so, you probably don't have to worry about the detail in your initial question at all. Instead, See the example at:
https://github.com/ahundt/grl/blob/master/test/AzmqFlatbufferTest.cpp

does that help?

Ashkan372, 2016-06-07:
Yes, awesome! it helped.
Although the details were also interesting, however not that important.

closed 2016-06-08 by Ashkan372