ahundt/grl #102: flatbuffers C++ segmentation fault

Ashkan372, 2016-06-28:
```
auto rbPstart = (const unsigned char *)message_rcv.data();
 auto verifier = flatbuffers::Verifier(rbPstart,message_rcv.size());
auto bufOK = grl::flatbuffer::VerifyKUKAiiwaStatesBuffer(verifier);
```

```
if (bufOk){

auto fbKUKAiiwaStates = flatbuffer::GetKUKAiiwaStates(rbPstart);
auto force = fbKUKAiiwaStates->states()->Get(0)->monitorState()->CartesianWrench()->force();
double z = force.z();
std::cout << "received force:" << z << "\n";

}
```
This is on the c++ side after you receive the msg with zmq... the last statement(cout) causes segmentation fault.
However I reopen the flatbuffer in the java side to check if the numbers are actually there, and they are really there!  also the buffer size is the same before sending and after receiving... the verification also looks good.. but don't know why the values are not inside in C++ side!

ahundt, 2016-06-29:
Hmm that is a bit puzzling. And code that receives a dummy test in a soul
at way works? Have you tried stepping through in a debugger to look at the
data?

Ashkan372, 2016-06-29:
No.. but I assume it is NULL..

ahundt, 2016-06-29:
Okay well I can't really see much from the code that's there And can't
test. I suggest writing a small bit of test code. Also You could submit a
pull request for me to look at.

Ashkan372, 2016-06-30:
Hi.. I checked in debugger.. and it is a random number like 2.230892394083e-317... although I had put 2.0 in this variable in flatbuffer... The strange thing is that the verifier works... but seems that the buffer contains junk... or something wrong with deserilization

Ashkan372, 2016-06-30:
Hey Andrew can you please help me debug how to decompose flatbuffers and read the data.

Can you please have a look on the way how I receive the data till the way I try to decompose the recv flatbuffer. Relevant code is 

https://github.com/Ashkan372/grl/blob/master/include/grl/kuka/KukaJAVAdriver.hpp#L417-L462

Thanks a lot.

Ashkan372, 2016-07-01:
There was a minor problem on Java side, this C++ code was fine.

closed 2016-07-01 by Ashkan372