ahundt/grl #105: Documentation for using grl with java and V-REP

apfohl, 2016-07-20:
Hello,

is there any documentation on how to use grl from a java application to control a simulation in V-REP?

ahundt, 2016-07-20:
You’d just set it up normally and follow the V-REP documentation for
controlling a simulation from Java.

Cheers!
Andrew Hundt

Ph.D. Student
Johns Hopkins University

apfohl, 2016-07-20:
That easy? Ok, I'll try.

ahundt, 2016-07-20:
Well that's the theory, it is untested and there are probably some
configuration changes that will be needed, plus you'll need to make sure
your scene is created to work with the name strings expected by the driver.

I also recently switched to having two simulated arms in one location for a
single physical arm in my simulation, one that reflects the sensors and
another that reflects what's being commanded.

closed 2016-10-26 by apfohl

ahundt, 2016-10-26:
@apfohl guess it didn't work out for you?

apfohl, 2016-10-26:
Yes, sorry. I had not much time to try and figure it out. I switched to Python for controlling the simulation. But grl is still a nice project!

ahundt, 2016-10-26:
cool, glad you found a good solution!