ahundt/grl #118: Bug in basis_tokenize_dependency in CommonTools.cmake

davidadley, 2017-01-31:
I am trying to compile with the sch-core dependency. However the basis_tokenize_dependency() function tokenises "sch-core" into a package name ("sch") and also a version number ("core"), this breaks cmake as it cannot find the pkgconfig file.

Line 119 in CommonTools.cmake has the regexp. I've simply commented out this if statement.

ahundt, 2017-01-31:
Thanks for the report! The version installed by the robotics_setup scripts contains a fix. I suggest taking a look at or following along with those setup scripts:

https://github.com/ahundt/robotics_setup/blob/master/cmake-basis.sh

This is the branch of cmake-basis with two fixes already merged:
https://github.com/ahundt/BASIS/tree/grl

Pull request for fix is here:
https://github.com/cmake-basis/BASIS/pull/609
Issue reporting the problem is here:
https://github.com/cmake-basis/BASIS/pull/608

I'll leave this open until the fixes are released upstream.

davidadley, 2017-01-31:
Many thanks for your quick response!

Problem solved.

davidadley, 2017-01-31:
I might have been a bit hasty above, I am now running into the same error as you did previously:

https://github.com/jrl-umi3218/Tasks/issues/6 #

pkg-config finds the package

`pkg-config --exists --print-errors sch-core`

returns ok

However cmake cannot as sch-core is not configured with the correct cmake files (sch-coreConfig.make or Findsch-core.cmake):

'cmake --find-package -DNAME=sch-core -DCOMPILER_ID=GNU -DLANGUAGE=CXX -DMODE=EXIST'

returns an error. 

How did you fix this? I tried pulling the ahundt/sch-core but same issue.

ahundt, 2017-01-31:
Whoops, I hadn't created a pull request for one fix:

https://github.com/jrl-umi3218/sch-core/pull/11

A script for installing Tasks and all its dependencies can be found here:
https://github.com/ahundt/robotics_setup/blob/master/robotics_tasks.sh

davidadley, 2017-01-31:
Many thanks,
That fixed the problem.

closed 2017-02-02 by davidadley