ahundt/grl #127: add a mode_service for changing interaction_mode

tdinesh, 2017-02-08:
Currently there is no std_srvs for sending a string. How do you want to structure non-standard srvs in grl? Thought on how to handle message_generation/message_runtime in CMake Basis?

ahundt, 2017-02-08:
sorry could you give additional context? Where is the string being sent from/to?

CMake basis just assists with compilation, are you discussing the generation of ROS messages?

In that case it is good to be aware of existing work to see if you could make use of that, specifically the messages available in [iiwa_stack](https://github.com/SalvoVirga/iiwa_stack/tree/master/iiwa_msgs/msg) and in [majorana](https://github.com/ros-industrial-consortium/majorana/issues/3).

I'm okay with adding new dependencies, I'd just like to keep them optional if it is sensible.

For generating messages or adding new cmake functions or importing external ones, that will typically go in [grl/config/Settings.cmake](https://github.com/ahundt/grl/blob/master/config/Settings.cmake), as you can see the flatbuffer config is there at the moment. Though perhaps for ros specific items the [grl/src/ros/CMakeLists.txt](https://github.com/ahundt/grl/blob/master/src/ros/CMakeLists.txt) may be appropriate.

tdinesh, 2017-02-08:
In addition to publishing std_msgs::String to switch the interaction mode, I would like to have a service which has a request field of String. Since "ArmState" is a custom enum, not sure any of the "iiwa_stack" or "majorana" srvs are usable.

ahundt, 2017-02-09:
I see, alternately for the enum perhaps requesting an integer which is convertible to the enum would be sensible? Up to you.