ahundt/grl #128: find_pacakge(grl), ${grl_LIBRARIES}

tdinesh, 2017-02-09:
I would like to link GRL libraries to a catkin package. grlflatbuffers targets does not get installed currently in /opt/grl/lib. I would like to have ${grl_LIBRARIES} be set for cmake. Could you point me where to set the basis install targets.

ahundt, 2017-02-09:
I suggest using `cmake -DCMAKE_INSTALL_PREFIX="/usr/local"` (in addition to the other build script parameters) to set it to install the project config files. Then run `make install` to install the grlConfig.cmake into `/usr/local/lib/cmake/grl`. You should see something like:

```bash
± ls /usr/local/lib/cmake/grl
grlConfig.cmake         grlExports.cmake          grlInstallManifest.txt  grlUse.cmake
grlConfigVersion.cmake  grlExports-release.cmake  grlUninstall.cmake
```

tdinesh, 2017-02-09:
Somehow it still does not set the correct includes. 
/usr/local/include/grl/kuka/KukaFRIdriver.hpp:22:27: fatal error: friClientData.h: No such file or directory

ahundt, 2017-02-09:
oh... I see the export install/public interface flags will need to be added to the [FRI cmake script](https://github.com/ahundt/grl/blob/master/config/FRI_Client_SDK_Cpp.cmake), and we will need to make sure the headers are installed to an appropriate location. This should be a fairly easy fix via cmake-basis or some standard cmake code. I just think most users went with ros messages or the vrep api up until now which doesn't require the includes.

This is the relevant include spec component:
https://cmake.org/cmake/help/v3.6/command/target_include_directories.html

Are the missing headers actually installed somewhere in /usr/local/include?

If not an install line for the relevant headers will need to be added as well. Nothing too complex but will involve some cmake code tweaks.

ahundt, 2017-02-09:
The above plus this function:

        basis_install_directory(FRI_Client_SDK_Cpp/src/protobuf_gen)

https://cmake-basis.github.io/apidoc/modules.html

It may also be worthwhile to create a quick issue on the cmake-basis github page https://github.com/cmake-basis/BASIS/issues

maybe this one in particular:
https://github.com/cmake-basis/BASIS/issues/339

tdinesh, 2017-02-09:
The missing headers are not installed in /usr/local/include.

ahundt, 2017-02-09:
Aha, so the super temporary fix would be to literally copy-paste the files there (or some other appropriate included directory). In the meantime we can work out the cmake patch that permanently fixes it.

ahundt, 2017-02-09:
created new basis issue about this:
https://github.com/cmake-basis/BASIS/issues/621

tdinesh, 2017-02-09:
After manually including the directories of missing files now I get linker error. 

`In function grl::robot::arm::decode(KUKA::FRI::ClientData&, unsigned long)':
main.cpp:(.text+0x83b): undefined reference to KUKA::FRI::MonitoringMessageDecoder::decode(char*, int)'
collect2: error: ld returned 1 exit status`

ahundt, 2017-02-09:
yeah you will need to link the FRI libraries and the protobufc that are compiled by the hyperlinked cmake file as well.

ahundt, 2017-02-09:
This is the cmake file I'm speaking of:
https://github.com/ahundt/grl/blob/master/config/FRI_Client_SDK_Cpp.cmake

ahundt, 2017-02-09:
so the libraries will be:
KukaFRIClient
friUdpConnection
nanopb

Which should at least be in the build folder, and hopefully we can patch things to put it in the install lib/include folders as well.

tdinesh, 2017-02-09:
${grl_LIBRARIES} is empty. The libs are present in /usr/local/lib/grl. I will manually link them. Thanks!

ahundt, 2019-08-26:
> I hope to use the grl in ROS project, but just using find package grl is
> not enough.
>
> find_package(grl)
>
> I get the following errors from cmake:
>
> undefined reference to `KUKA::FRI::MonitoringMessageDecoder::decode(char*, int)'

It looks like flatbuffers isn’t running correctly to generate that code.
Check that the flatbuffers generator is being run

jacknlliu, 2019-08-26:
I hope to use the `grl` in ROS project, but just using find package grl is not enough.

```
find_package(grl)
``` 

I get the following errors from cmake:
```
undefined reference to `KUKA::FRI::MonitoringMessageDecoder::decode(char*, int)'
```