ahundt/grl #138: IIWA KUKA programming in C++ without using ROS

sunnybau, 2017-09-10:
Respected Sir,
Is it possible that I can command IIWA KUKA in C++ language by using matlab without using ROS.
I read your below link where you mention that you've written a set of libraries for operating the kuka iiwa from C++ without dependencies on ROS.
 https://groups.google.com/forum/#!topic/swri-ros-pkg-dev/8ykHemP8CHY

ahundt, 2017-09-12:
You are correct that grl requires C++ and does not require ROS. It is
possible to control V-REP from MATLAB, and V-REP is able to control the
KUKA IIWA via grl. Therefore, it may already be possible to control the
KUKA from MATLAB, but this has not been tested.

If you do not wish to use V-REP, it is also perfectly possible to write a
MATLAB interface to the KUKA separately using grl, but this is not
implemented.

Cheers!
Andrew Hundt

Ph.D. Student
Johns Hopkins University

ahundt, 2017-09-21:
closing due to inactivity, feel free to comment again if you have any other questions

closed 2017-09-21 by ahundt