ahundt/grl #152: Position Control using ROS

Pyrojambo, 2017-12-07:
I am working on a project where I need to directly command Cartesian or joint positions to a KUKA iiwa LBR 14 using ROS. These positions are calculated in C++ from force sensor readings at either 500 or 1000Hz. I have previously tried this using DirectServo and SmartServo without using a FRI connection but the movements were performed too slowly.
Is there functionality in your toolkit that allows for this?
Thanks.

ahundt, 2017-12-07:
It is possible to control joint positions over FRI with GRL, and we've been able to command positions. Do you need specific instructions on activating FRI with GRL? Have you been able to install the driver on Sunrise and command the arm with GRL? If so activating the FRI API should be fairly simple and we can go from there.  Please note the FRI feature is still a work in progress but we are happy to work with you on it.

Regarding the force and torque sensors in the arm, there are known issues with the iiwa LBR and the readings are very inaccurate. A solution to this problem has recently been published, and we'd love to support integration of it into GRL. See https://www.disneyresearch.com/publication/toward-torque-control-of-a-kuka-lbr-iiwa/ for details, the paper itself can also be downloaded at that link. Note that this would be a substantial undertaking, and if you'd like a quick-fix solution I recommend simply purchasing a force torque sensor such as those provided by robotiq or other firms and attaching it to your end effector.

Pyrojambo, 2017-12-08:
I have installed the driver on the Sunrise controller so a brief how-to on how to use FRI with grl and which Java APIs and cpp files work together would be most useful.

I am currently using an external force sensor attached to the end eff for this project, so will be sticking with the simple option for now.

ahundt, 2017-12-08:
Be very careful the first time you run things with the big read button handy at all times. The robot should start straight up with all joints at 0 for this test.

So you followed the following steps?
https://ahundt.github.io/grl/howto/iiwaKukaRobotSetup.html

Once those steps are all done, starting up the driver is just selecting GRL_Driver from the kuka teach pendant's application list, ensuring autonomous mode is enabled and hit play. It will sit an wait there until grl connects to it.

If you would like to test the ROS driver, you just launch the driver executable `grl_kuka_ros_driver` once it it built (make sure you have it enabled & building with CMake), and also make sure the robot itself is waiting for a connection.

Then run the [kuka_ros_test_command.sh](https://github.com/ahundt/grl/blob/master/src/ros/kuka_ros_test_command.sh) to make it move, just one joint should rotate a few degrees. You can play around and modify the script to send a variety of joint positions.

```
# To run a test driving the kuka iiwa into the straight up position,
# launch each of the following in separate terminals:
# roscore 
# grl_kuka_ros_driver
# rostopic pub /interaction_mode std_msgs/String "data: 'MoveArmJointServo'"
rostopic pub /joint_traj_pt_cmd trajectory_msgs/JointTrajectoryPoint "positions: [0,0,0,0,0,0,0.1]
velocities: [0,0,0,0,0,0,0]
accelerations: [0,0,0,0,0,0,0]
effort: [0,0,0,0,0,0,0]
time_from_start: {secs: 1, nsecs: 0}"
```

We also have a fully integrated ROS setup that works with the kuka and integrates with https://github.com/IFL-CAMP/iiwa_stack/ at:
https://github.com/cpaxton/costar_stack/

ahundt, 2017-12-19:
Do you have any other questions?

Pyrojambo, 2017-12-19:
Sorry, I've had to work on other things instead recently.
I'm now trying to build the grl_kuka_ros_driver.cpp using catkin_make but it doesn't build any of cpp files

ahundt, 2017-12-19:
you need to use ccmake and set `WITH_ROS=ON`, then it will build the ROS node.

Alternately you can see how it is done in:
https://github.com/ahundt/robotics_setup/blob/master/grl_kuka.sh

Don't forget to put the FRI zip in the data folder or the driver won't build.

Pyrojambo, 2017-12-20:
I'm still having issues with `ccmake ..` and `make -j`. The errors seem to come from .fbs files not building their header files.
Also the FRI zip folder I have doesn't have the two TransformationProviderApp folders, will this also be causing errors?

ahundt, 2017-12-21:
which version of grl and sunrise are you using?

Pyrojambo, 2018-01-03:
I am using the latest version of grl on the master branch. Sunrise version is 1.10.0.8

ahundt, 2018-01-03:
Ah we have 1.11, and I think that might be the source of the problem. Is there any chance it might not be too difficult to upgrade?

Pyrojambo, 2018-01-17:
I managed to get Sunrise 1.11 which solved the issues with the TransformationProviderApp folders.
But I still have issues during `make -j`, when it converts fbs into header files then following message appears a lot:
`[ 36%] Building C++, Java, and Python header for Time.fbs`
`/bin/sh: 1: --gen-name-strings: not found`

ahundt, 2018-01-19:
ah seems like you need to update your flatbuffers version as well. The current release is https://github.com/google/flatbuffers/tree/v1.8.0

Pyrojambo, 2018-01-19:
It was 1.8 that I had but reinstalled it from that link and still got the same problem.
`[ 13%] Building C++, Java, and Python header for Time.fbs`
`/bin/sh: 1: --gen-name-strings: not found`
`CMakeFiles/grlflatbuffers.dir/build.make:188: recipe for target 'include/grl/flatbuffer/Time_generated.h' failed`
`make[2]: *** [include/grl/flatbuffer/Time_generated.h] Error 127`
`make[2]: *** Waiting for unfinished jobs....`

ahundt, 2018-01-19:
sorry can you run with `make VERBOE=1` then I can see the actual command line commands

Pyrojambo, 2018-01-23:
`[ 73%] Building C++, Java, and Python header for LogKUKAiiwaFusionTrack.fbs`
`cd /home/jamie/src/grl/include/grl/flatbuffer && --gen-name-strings --scoped-enums --gen-object-api -c -j -p -o /home/jamie/src/grl/build/include/grl/flatbuffer LogKUKAiiwaFusionTrack.fbs`
`/bin/sh: 1: --gen-name-strings: not found`
`CMakeFiles/grlflatbuffers.dir/build.make:92: recipe for target 'include/grl/flatbuffer/LogKUKAiiwaFusionTrack_generated.h' failed`
`make[2]: *** [include/grl/flatbuffer/LogKUKAiiwaFusionTrack_generated.h] Error 127`
`make[2]: Leaving directory '/home/jamie/src/grl/build'`
`CMakeFiles/Makefile2:483: recipe for target 'CMakeFiles/grlflatbuffers.dir/all' failed`
`make[1]: *** [CMakeFiles/grlflatbuffers.dir/all] Error 2`
`make[1]: Leaving directory '/home/jamie/src/grl/build'`
`Makefile:160: recipe for target 'all' failed`
`make: *** [all] Error 2`

ahundt, 2018-01-25:
very strange, && is being inserted between... The command should be something more like:

`flatc --gen-name-strings --scoped-enums --gen-object-api -c -j -p -o /home/jamie/src/grl/build/include/grl/flatbuffer/LogKUKAiiwaFusionTrack.fbs`

The flatc command generates code that helps with sending messages to the robot.

Which version of grl are you on, master or 4.1?
I'll try to reproduce it

Pyrojambo, 2018-01-25:
I'm using the master branch version

ahundt, 2018-01-25:
I'm unable to reproduce the error, things seem to compile ok for me, including flatc. What version of cmake are you using?

Here is mine:
```
± cmake --version
cmake version 3.7.1
```

Pyrojambo, 2018-01-26:
CMake version is 3.5.1
Updating to the latest version and will see what happens

Pyrojambo, 2018-01-26:
Updated to 3.10.2 but still getting the same result. When I run the `cmake ..` section of grl_kuka.sh with documentation off (because for some reason it can't find `sphinx-build` even though it is installed) I get the following:
[grl cmake results.txt](https://github.com/ahundt/grl/files/1668304/grl.cmake.results.txt)

Leytha, 2019-07-30:
Probably too late, but I ran into the same errors (flatc one and sphinx one) and after some tries, I managed to correctly compile the code:
- Completely remove the /build dir inside /grl (make clean was not enough)
- Create again the /build dir and run ccmake -Wno-dev ..  (the -Wno-dev flag is because in the newer versions of CMake it asks for a line containing project() and it returns an annoying warning, but you don't have to put it if you don't want to)
- Disable the option BUILD_DOCUMENTATION (because I don't know what's happening with Sphinx but I also have the same issue and it doesn't find it)
- Disable TRTK because when I download it from source and try to compile it I get several errors I'm not able to fix
- Reconfigure and generate (c and g)
- Run "make", without -jwhatever. I don't exactly understand why it doesn't work when compiling with several cores, but with just one it does compile. 

I also had to change two files due to another error with spdlog:
- ~/src/grl/include/grl/kuka/KukaJAVAdriver.hpp
- ~/src/grl/src/v_repExtHandEyeCalibration/v_repExtHandEyeCalibration.cpp.

I had to add at the top of both of them this include line:
`#include <spdlog/sinks/stdout_sinks.h>`

Otherwise I got the following error several times in different parts of the mentioned files:
> error: ‘stdout_logger_mt’ is not a member of ‘spdlog’

ahundt, 2019-08-09:
thanks! Sorry that I had not seen these posts until now

ahundt, 2019-08-09:
Some of these items might be fixed in https://github.com/ahundt/grl/pull/166

jacknlliu, 2019-08-24:
@ahundt  did you use the [iiwa_ros_java](https://github.com/IFL-CAMP/iiwa_stack/tree/master/iiwa_ros_java) on the KUKA workbench project when you integrate the [iiwa_stack](https://github.com/IFL-CAMP/iiwa_stack)?

ahundt, 2019-08-24:
I’m trying to recall the details, I think I ran the GRL Java driver on the
robot but integrated the models from iiwa stack and possibly a couple of
rod messages. There is a small ros program built into GRL which converts
everything to ros messages. You might have to check the cmake settings and
dependencies to enable it.