ahundt/grl #175: GRL import on Sunrise Workbench

kamilcetin, 2018-11-21:
I have some problem on importing grl project on Sunrise Workbench.
I followed the instructions of Kuka LBR iiwa Java Setup at https://ahundt.github.io/grl/howto/iiwaKukaRobotSetup.html

I got stuck with the step of Setup Java Driver Software. I copy the folders in ${GRL_DIR}/src/java/grl/ from Ubuntu 16.04, then try to import the grl project with the Sunrise Workbench. However, when I select the grl folder from File > Import > General > Existing Projects into Sunrise Workspace GUI, the project does not seem on the list and it is not possible to Finish as shown below.
[image]

On the other side, when I import the grl project from File > Import > General > File System, it is loaded on the Sunrise Project. In this case, it is not possible to install the SunriseProject with imported files of grl onto the Kuka Controller due to the errors on the grl, grl.driver, grl.flatbuffer, grl.flatBufferTesting as shown below.
[image]

Do you have an idea for the reason of the problem? Is the way of importing grl on the SunsireProject true? Should I install more packages on the project to prevent the errors at ...java files? 
Thanks in advance.

ahundt, 2018-11-21:
I think you might be able to simply manually add the files by creating a
fresh project and copy them in. I'm traveling at the moment so I might not
be able to respond immediately.

ahundt, 2018-11-23:
flatbuffers and zmq are actual java libraries you need to install here is the flatbuffers one, for example:

https://github.com/davidmoten/flatbuffers

kamilcetin, 2018-11-26:
I found those java libraries and directly copied into the src directory of the project in the interface. Then I could install and do the sync. Then I changed the Ethernet port to FRI/Koni and saw GRL_driver application from the smart pad. When I chose it and run, the interface says that FRI Joint Overlay Complete! I think I completed the FRI and GRL installations on both side (remote ubuntu and KRC worksation).
Now, where can I find example cpp codes and how to run them from ROS?

ahundt, 2018-12-03:
"FRI Joint Overlay Complete!" is printed when everything is initialized, that means it is working. I know the wording is unclear I just never got around to fixing it, sorry about that.

Once you compile following the GRL instructions the cpp code is in  https://github.com/ahundt/grl/tree/master/src/ros there is also an example shell script.

kamilcetin, 2018-12-12:
It works now!
When I connect both ethernet ports of the Kuka to my computer with two separate ethernet ports, I could run the grl.driver and move the robot through FRI.

ahundt, 2018-12-13:
cool! I'm super happy to hear that! 👍 @cpaxton

kamilcetin, 2018-12-14:
About the GRL installation, I have no trouble. 

Now, I am dealing with the inverse kinematics solutions of Kuka lbr iiwa. Do you know which src codes in grl repository I can use or where I have to look at?

ahundt, 2018-12-14:
IK can be a bit painful to set up because it depends on each particular robot's attachments etc. 

I use tasks for IK & control https://github.com/jrl-umi3218/Tasks.

I created a model of my robot in V-REP, and created a plugin in [InverseKinematicsVrepPlugin.hpp](https://github.com/ahundt/grl/blob/master/include/grl/vrep/InverseKinematicsVrepPlugin.hpp) which loads the specific joints in my model. In this plugin the geometry is copied from V-REP into Tasks, and then I use the tasks solver to do the planning.

kamilcetin, 2018-12-17:
In order to make easy for me, first, I think I should start to use its Jacobian matrix from the forward kinematic equations. How can I create the kinematic model of Kuka lbr iiwa 14 in V-REP?

ahundt, 2018-12-17:
It's in the list of existing robot models in the app. You just drag and
drop. I think there is an example vrep scene mentioned in the documentation
you can use.

kamilcetin, 2018-12-17:
Which one is the example what you mentioned; [v_repExtPluginSkeleton.cpp](https://github.com/ahundt/grl/blob/master/example/v_repExtPluginSkeleton/v_repExtPluginSkeleton.cpp) or [KukaLBRiiwaVrepPluginTest.cpp](https://github.com/ahundt/grl/blob/master/test/KukaLBRiiwaVrepPluginTest.cpp) ?

ahundt, 2018-12-19:
the .ttt files in https://github.com/ahundt/grl/tree/master/data can be opened with http://coppeliarobotics.com/index.html. There might be a newer version in https://github.com/ahundt/robone/tree/master/data, or I might have to update it.

Usually I have one model that tracks where the arm is now, and another model which tracks where the arm will go. I put them both at the same base position in the same simulation. FYI these kinematics features are ones we've run on multiple robots but I'm not sure anyone outside our lab has integrated them via GRL yet. All of the components are tested on multiple robots but configuration may have a steep learning curve.

ahundt, 2019-12-27:
Closing this issue because it has not been updated in a while. Feel free to reopen

closed 2019-12-27 by ahundt