ahundt/grl #178: ros test command not work

jacknlliu, 2019-08-27:
Hi, after some hard work, the coding is finally running both on SmartHMI  and PC, but  when I test the [kuka ros  test script](https://github.com/ahundt/grl/blob/master/src/ros/kuka_ros_test_command.sh), the robot didn't move to the specified joint position.

the output from PC is:
```
 [console] [info] re-extracted 0 joint angles: []
```

the output from SmartHMI
```
Didn't receive correct number of joints! skipping to start of loop...
```

I set the params as the following list, and I only connect the x66 port on the KUKA controller box.
```
RobotName: "Robotiiwa"
RobotModel: "KUKA_LBR_IIWA_7_R800"
LocalUDPAddress: "0.0.0.0"
LocalUDPPort: "30010"
RemoteUDPAddress: "172.31.1.160"
LocalHostKukaKoniUDPAddress: "192.168.1.100"
LocalHostKukaKoniUDPPort: "30200"
RemoteHostKukaKoniUDPAddress: "192.168.1.2"
RemoteHostKukaKoniUDPPort: "30200"
KukaCommandMode: "JAVA"
KukaMonitorMode: "JAVA"
```

jacknlliu, 2019-08-27:
@ahundt  I think there is a mistake here

https://github.com/ahundt/grl/blob/bd62058cb3d7fdaf85ea2841d32199389152e7fa/include/grl/ros/KukaLBRiiwaROSPlugin.hpp#L93

here get the `LocalUDPPort` param from ros, but assigned it to `LocalUDPAddress`!

jacknlliu, 2019-08-27:
After fix the param, also not working
```
[ INFO] [1566872228.070443406]: done creating subscribers
[2019-08-27 02:17:08.070] [console] [info] KukaLBRiiwaRosPlugin: Connecting UDP Socket from 
[ INFO] [1566872228.070756127]: KukaLBRiiwaROSPlugin in unsupported mode! Valid grl::flatbuffer::ArmState required for interaction_mode topic!
[*** LOG ERROR #0001 ***] [2019-08-27 02:17:08] [console] {argument index out of range}
[*** LOG ERROR #1002 ***] [2019-08-27 02:17:09] [console] {argument index out of range}
[*** LOG ERROR #2002 ***] [2019-08-27 02:17:10] [console] {argument index out of range}
[*** LOG ERROR #3003 ***] [2019-08-27 02:17:11] [console] {argument index out of range}
[2019-08-27 02:17:11.733] [console] [error] C++ KukaJAVAdriver Error: flatbuff failed verification. bufOk: false
[*** LOG ERROR #4004 ***] [2019-08-27 02:17:12] [console] {argument index out of range}
[*** LOG ERROR #5005 ***] [2019-08-27 02:17:13] [console] {argument index out of range}
[*** LOG ERROR #6006 ***] [2019-08-27 02:17:14] [console] {argument index out of range}
[*** LOG ERROR #7006 ***] [2019-08-27 02:17:15] [console] {argument index out of range}
[*** LOG ERROR #8007 ***] [2019-08-27 02:17:16] [console] {argument index out of range}
[*** LOG ERROR #9007 ***] [2019-08-27 02:17:17] [console] {argument index out of range}
[ INFO] [1566872238.070844170]: KukaLBRiiwaROSPlugin in unsupported mode! Valid grl::flatbuffer::ArmState required for interaction_mode topic!
[*** LOG ERROR #10008 ***] [2019-08-27 02:17:18] [console] {argument index out of range}
[*** LOG ERROR #11008 ***] [2019-08-27 02:17:19] [console] {argument index out of range}
[*** LOG ERROR #12009 ***] [2019-08-27 02:17:20] [console] {argument index out of range}
[*** LOG ERROR #13010 ***] [2019-08-27 02:17:21] [console] {argument index out of range}
[*** LOG ERROR #14011 ***] [2019-08-27 02:17:22] [console] {argument index out of range}
[*** LOG ERROR #15011 ***] [2019-08-27 02:17:23] [console] {argument index out of range}
[*** LOG ERROR #16011 ***] [2019-08-27 02:17:24] [console] {argument index out of range}
[*** LOG ERROR #17012 ***] [2019-08-27 02:17:25] [console] {argument index out of range}
[*** LOG ERROR #18013 ***] [2019-08-27 02:17:26] [console] {argument index out of range}
[*** LOG ERROR #19013 ***] [2019-08-27 02:17:27] [console] {argument index out of range}
[ INFO] [1566872248.070828425]: KukaLBRiiwaROSPlugin in unsupported mode! Valid grl::flatbuffer::ArmState required for interaction_mode topic!
[*** LOG ERROR #20013 ***] [2019-08-27 02:17:28] [console] {argument index out of range}
[*** LOG ERROR #21014 ***] [2019-08-27 02:17:29] [console] {argument index out of range}
[*** LOG ERROR #22015 ***] [2019-08-27 02:17:30] [console] {argument index out of range}
[*** LOG ERROR #23015 ***] [2019-08-27 02:17:31] [console] {argument index out of range}
[*** LOG ERROR #24016 ***] [2019-08-27 02:17:32] [console] {argument index out of range}
[*** LOG ERROR #25017 ***] [2019-08-27 02:17:33] [console] {argument index out of range}
[*** LOG ERROR #26018 ***] [2019-08-27 02:17:34] [console] {argument index out of range}
[*** LOG ERROR #27018 ***] [2019-08-27 02:17:35] [console] {argument index out of range}
[*** LOG ERROR #28018 ***] [2019-08-27 02:17:36] [console] {argument index out of range}
[*** LOG ERROR #29019 ***] [2019-08-27 02:17:37] [console] {argument index out of range}
[ INFO] [1566872258.070853638]: KukaLBRiiwaROSPlugin in unsupported mode! Valid grl::flatbuffer::ArmState required for interaction_mode topic!
[*** LOG ERROR #30019 ***] [2019-08-27 02:17:38] [console] {argument index out of range}
```

jacknlliu, 2019-08-27:
Another problem, why is there a `tcp` connection `tcp://xxx:30010` on the smartpad, actually we use the udp socket?

jacknlliu, 2019-08-27:
the robot can move using the kuka ros test script, but the `/joint_states` can't get the valid message.

The following output from the grl kuka driver:
```
[2019-08-27 13:36:52.073] [console] [error] C++ KukaJAVAdriver Error: flatbuff failed verification. bufOk: false
[2019-08-27 13:36:52.077] [console] [error] C++ KukaJAVAdriver Error: flatbuff failed verification. bufOk: false
[2019-08-27 13:36:52.081] [console] [error] C++ KukaJAVAdriver Error: flatbuff failed verification. bufOk: false
[2019-08-27 13:36:52.086] [console] [error] C++ KukaJAVAdriver Error: flatbuff failed verification. bufOk: false
[2019-08-27 13:36:52.090] [console] [error] C++ KukaJAVAdriver Error: flatbuff failed verification. bufOk: false
[2019-08-27 13:36:52.093] [console] [error] C++ KukaJAVAdriver Error: flatbuff failed verification. bufOk: false
[2019-08-27 13:36:52.107] [console] [error] C++ KukaJAVAdriver Error: flatbuff failed verification. bufOk: false
[2019-08-27 13:36:52.118] [console] [error] C++ KukaJAVAdriver Error: flatbuff failed verification. bufOk: false
[2019-08-27 13:36:52.131] [console] [error] C++ KukaJAVAdriver Error: flatbuff failed verification. bufOk: false
[2019-08-27 13:36:52.135] [console] [error] C++ KukaJAVAdriver Error: flatbuff failed verification. bufOk: false
[2019-08-27 13:36:52.139] [console] [error] C++ KukaJAVAdriver Error: flatbuff failed verification. bufOk: false
```

The following output from the SmartHMI:
[image]

jacknlliu, 2019-08-29:
After fixing the java side, the verification error disappeared. But the topic `joint_states` and `/slave/wrench_r` is still not available. Then I find the new data flag is always false when the Java monitor mode is used, and I change the flag then the `/slave/wrench_r` and `/joint_states` are published.  But the `/joint_states` return empty positions. 

I think the default usage for this repo is that using the JAVA interface as the command interface but use the FRI as the monitor interface, is it right? @ahundt   If this is the default behavior, how to set the correct control mode both for FRI and SmartServo interface?

ahundt, 2019-09-10:
the default is to use java + smartservo.

you need to enable the ros integration in cmake, and run that for stuff to appear on ROS.

ahundt, 2019-09-10:
This is the relevant cpp file you should make sure is compiled and run:
```
https://github.com/ahundt/grl/blob/master/src/ros/grl_kuka_ros_driver.cpp
```

This is the relevant cmake file:
```
https://github.com/ahundt/grl/blob/master/src/ros/CMakeLists.txt
```

Make sure the ccmake `*_FOUND` flags are all enabled.

jacknlliu, 2019-09-11:
@ahundt thanks, these things I have completed. I debug them and find some bugs like the as described above, and I think the disadvantage of this driver is that the ros control hardware interface is not implemented. And the default usage of this driver is not intuitive and feature-rich as [iiwa stack](https://github.com/IFL-CAMP/iiwa_stack).

ahundt, 2019-12-27:
This API is designed for soft realtime use and it includes its own controller so ros control was never needed for our use cases. Similarly, Java wrench measurement calls add latency. If iiwa stack fits your project's needs, you should use it!

Thanks for giving this a try!

closed 2019-12-27 by ahundt