ahundt/grl #180: FlatBuffers Library not Found

danielj195, 2020-02-10:
Hi,

When I try to install your code from source, I get a message saying that "Project grl requires FlatBuffers.  Please ensure that the package is installed in a standard system location".  I've tried installing Flatbuffers from source and using snap binaries.  Here are the instructions I followed to install it in my system folder:
https://stackoverflow.com/questions/55394537/how-to-install-flatc-and-flatbuffers-on-linux-ubuntu/55394568#55394568

When I recompile, I am still getting this error, so I am not sure what to do at this point.  I would be very grateful if you could point me in the right direction.  Thank you!

-Daniel

ahundt, 2020-02-10:
I’m very sorry I don’t have time to support this project over the next
couple months. CMake looks in certain directories, which you can find
specified in the CMake files.

I’m sorry if that isn’t very helpful, but that is all I have time for at
the moment.

matthias-mayr, 2020-06-15:
I used the instructions in https://github.com/ahundt/grl/blob/master/INSTALL.md with 18.04
```
git clone https://github.com/ahundt/robotics_setup.git
cd robotics_setup
./grl_kuka.sh
```
I do not remember that I had issue with flatbuffers when compiling it on the Linux machine.

matthias-mayr, 2020-06-15:
**If people are here because they have issues with flatbuffers on the Java/Sunrise side:**

- I downloaded the jars from [Maven](https://mvnrepository.com/artifact/com.google.flatbuffers/flatbuffers-java)
- For me with Sunrise 1.15 it was version 1.8 that worked. Sunrise complains if it's the Java versions do not match by saying `com/google/flatbuffers/Table: Unsupported major.minor version 52.0`
- Download all the three `jar` files and place them in the folder `Third Party Libraries`
- Do a right click on them and selected `Build Path --> Add to build path`
- You might need to remove other version from your build path by doing another right click and selecting `Build Path --> Configure build path` 

@ahundt: What about pull requests for updates to the documentation and you commenting on them? Do you have time to review and merge them?

ahundt, 2020-06-25:
If there are any corrections that should be made, could you add them to a pull request?